tomatensaus/DeyeSolarDesktop #2: Lan connection

terzo33, 2024-04-15:
Hi @tomatensaus, your dongle is available with lan port for lan connection with my router? Or can I use esp32 with lan port?

Thanks

tomatensaus, 2024-04-17:
There is no reason to use LAN, the wifi always reconnects and work really well even if the signal is not so great. I have a prototype with a LAN8720 chip but there is no real reason to spend money to manufacture these if the current solution works well.

terzo33, 2024-04-17:
@tomatensaus Thanks a lot for the answer.  Unfortunately in my home due to some external interference the wifi is often disturbed and nothing works anymore.  I have already tried several routers without ever solving the problem.  cost is not a problem.  can you please develop the dongle with LAN socket.  

thank you infinitely

tomatensaus, 2024-04-23:
I will revive my LAN8720 project at some stage. There are some industrial installs where wifi is not available at all.

I am currently working on replacing the standard factory logger firmware with my own firmware. Got the first one working.

terzo33, 2024-04-23:
> I will revive my LAN8720 project at some stage. There are some industrial installs where wifi is not available at all.
> 
> I am currently working on replacing the standard factory logger firmware with my own firmware. Got the first one working.
> 
> 
>  
> 

Thanks a lot 🙏🏻🙏🏻

tomatensaus, 2024-08-20:
https://github.com/user-attachments/assets/471c65d8-424a-447b-a7a3-8c91f1935cd0

I managed to build a proof  of concept using the LAN8720
100Mbps LAN connection

Currently working on a new Dongle that supports both RS232 and RS485

terzo33, 2024-08-20:
Sounds good 😍😍 I'm waiting your news. Thanks a lot

urosgruber, 2024-11-18:
I would also love the LAN version. A wired connection is always better than wifi.