tomatensaus/DeyeSolarDesktop #3: will your dongle block interferance

iduqueno, 2024-04-22:
currently use standard esp32 and 485 board home assistant which works ok ish some errors but data ok i don't level shift or other stuff
but when  I grid  charge over night to the  battery even at low 50A or so it just falls over and reports junk suspect iv only got it just about working and the noise it creates doing this makes it fall over. Just looking for a little reassurance before I order

tomatensaus, 2024-04-23:
please show me an example of what you consider interference? Is this on the WIFI or RS485? I track the wifi signal on my loggers and here is an example of the wifi logger being disturbed from external interference. This history is on a system that charges every night from the grid to take advantage of cheap rates, it also discharges on peak rates 
[image]

iduqueno, 2024-04-23:
Hi thanks for the reply the problem im having from the logs is CRC Check fails i have a few normaly dont thinck my code is correct then i get a lot more  when charging till it crashes out
Any way thanks for response i will order your HW / software solution and see what happens

closed 2024-04-23 by iduqueno