tomatensaus/DeyeSolarDesktop #5: First tab "Deye" totally blank

bsafh, 2024-06-16:
`This is what the empty desktop looks like before any solar data is populated from the inverter.`

No, it's totally blank. Nothing there.

Even after installing everything from 
[JustGrabbingSomeIdeas](https://github.com/tomatensaus/DeyeSolarDesktop/blob/main/JustGrabbingSomeIdeas.md)
and replacing the entities for most of the cards (names are different in my Installation, as I use the Sunsyk/Deye addon with modbus connection) incl. all entities for the Sunsynk-Power-Flow-Card the tab stays just blank.

The Sunsynk-Power-Flow-Card works finde in another dashboard, but herein not even this shows up.

No errors displayed, no messing entities, nothing.

On the other tabs All the messing entities do show, but not on the first tab.

[image]

[image]

```
title: Solar
views:
  - theme: Backend-selected
    title: Deye
    path: deye
    subview: false
    cards:
      - type: custom:flex-horseshoe-card
        view_layout:
          grid-area: g2
        entities:
          - entity: sensor.wr_deye_1_battery_1_soc
            decimals: 0
            unit: '%'
            name: BATTERY
          - entity: sensor.wr_deye_1_battery_1_voltage
            decimals: 2
            unit: V
          - entity: ssensor.wr_deye_1_battery_1_current
            decimals: 2
            unit: A
          - entity: sensor.wr_deye_1_battery_1_power
            decimals: 0
            unit: W
          - entity: sensor.deye_1_day_battery_discharge
            decimals: 2
            unit: kWh
            name: Discharge
          - entity: ssensor.deye_1_day_battery_charge
            decimals: 2
            unit: kWh
            name: Charge
        show:
          horseshoe_style: autominmax
        layout:
          hlines:
            - id: 0
              xpos: 50
              ypos: 40
              length: 70
              styles:
                - opacity: 0.2;
            - id: 0
              xpos: 50
              ypos: 60
              length: 70
              styles:
                - opacity: 0.2;
          vlines:
            - id: 0
              xpos: 50
[... 738 lines cut ...]
            name: ' Generator Peak Shaving'
          - entity: number.deyeinvertermaster_generator_peak_shaving_power
            name: Generator Peak shaving power
          - entity: sensor.deyeinvertermaster_generator_port_power
            name: Generator Port Power
          - entity: binary_sensor.deyeinvertermaster_generator_connected_status
          - entity: switch.deyeinvertermaster_force_generator
            name: Force Generator
          - entity: switch.deyeinvertermaster_battery_generator_charge_enabled
            name: Generator Charge Enabled
          - entity: select.deyeinvertermaster_generator_signal
            name: Generator Signal
          - entity: number.deyeinvertermaster_generator_run_time
            name: Generator Run Time
          - entity: number.deyeinvertermaster_generator_down_time
            name: Generator Down Time
        title: Aux Port/Generator
        show_header_toggle: false
      - type: entities
        entities:
          - entity: sensor.deyeinvertermaster_inverter_time
          - entity: select.deyeinvertermaster_inverter_basic_settings_beep
          - entity: select.deyeinvertermaster_inverter_basic_settings_sync_clock
          - entity: button.deyeinvertermaster_inverter_sync_clock
```

bsafh, 2024-06-16:
This is the Sunsynk-Power-Flow-Card in another dashboard 

[image]

tomatensaus, 2024-06-17:
I suspect that you replaced an entity name that broke the yaml parsing. Use an editor such as VSCode, perhaps it can show you the invalid yaml

bsafh, 2024-06-17:
Might have been that .. after an unintentional reboot of the HA VM, something does show up, but the layout is completely broken (I never worked with the custom:layout-card):

The first to cards from do not show up, it starts with number 4, and the first card is HUGE, but on the right side and half behind card 7 ....

[image]
[image]

and scaling the whole screen down ("Ctrl" + "-" on Win10, Firefox) leads to totally bizarre results:

[image]

How to fix this?

```
title: Solar
views:
  - theme: Backend-selected
    title: Deye
    path: deye
    subview: false
    cards:
      - type: custom:flex-horseshoe-card
        card_height: 410px
        view_layout:
          grid-area: g2
        entities:
          - entity: sensor.wr_deye_1_battery_1_soc
            decimals: 0
            unit: '%'
            name: BATTERY
          - entity: sensor.wr_deye_1_battery_1_voltage
            decimals: 2
            unit: V
          - entity: sensor.wr_deye_1_battery_1_current
            decimals: 2
            unit: A
          - entity: sensor.wr_deye_1_battery_1_power
            decimals: 0
            unit: W
          - entity: sensor.deye_1_day_battery_discharge
            decimals: 2
            unit: kWh
            name: Discharge
          - entity: sensor.deye_1_day_battery_charge
            decimals: 2
            unit: kWh
            name: Charge
        show:
          horseshoe_style: autominmax
        layout:
          hlines:
            - id: 0
              xpos: 50
              ypos: 40
              length: 70
              styles:
                - opacity: 0.2;
            - id: 0
              xpos: 50
              ypos: 60
              length: 70
              styles:
                - opacity: 0.2;
          vlines:
            - id: 0
              xpos: 50
              ypos: 50
              length: 18
              styles:
                - opacity: 0.2;
[... 695 lines cut ...]
          - entity: sensor.deye_1_day_battery_discharge
            name: Battery
            color: rgb(151, 90, 182)
          - entity: sensor.deye_1_day_grid_import
            name: Grid
            color: rgb(84, 144, 194)
      - type: history-graph
        entities:
          - entity: sensor.wr_deye_1_load_power
          - entity: sensor.deye_1_grid_power
          - entity: sensor.wr_deye_1_pv_power
          - entity: sensor.wr_deye_1_battery_1_power
          - entity: sensor.deye_1_pv1_power
          - entity: sensor.deye_1_pv2_power
        hours_to_show: 12
      - type: weather-forecast
        entity: weather.dwd_uetze_e688_instanz_3_uetze_3
        forecast_type: daily
    visible:
      - user: 52d2c1ee572240deb997ebd3e5173a6d
      - user: 726dca442e764fbd98e216f638242af3
      - user: aded4704d04b4bf3b0256e41c83dfd54
    type: custom:grid-layout

```

bsafh, 2024-06-17:
`type: custom:grid-layout` does show the described behavior.

changing this to `type: custom:horizontal-layout` or Mansory does show all the cards ... so there must be some error in the layout definition (copied from [JustGrabbingSomeIdeas](https://github.com/tomatensaus/DeyeSolarDesktop/blob/main/JustGrabbingSomeIdeas.md) ).

Execpt for one card, which shows up but is empty - another wrong layout definition?

[image]

```

type: custom:flex-horseshoe-card
view_layout:
  grid-area: g1
  show: always
entities:
  - entity: sensor.wr_deye_1_pv_power
    decimals: 0
    unit: W
    name: Solar
  - entity: sensor.deye_1_pv1_power
    decimals: 0
    unit: W
    name: Output
  - entity: sensor.wr1_summary_day_pv1
    decimals: 1
    unit: kWh
    name: Produced
  - entity: sensor.deye_1_pv2_power
    decimals: 0
    unit: W
  - entity: sensor.wr1_summary_day_pv2
    decimals: 1
    unit: kWh
layout:
  hlines:
    - id: 0
      xpos: 50
      ypos: 40
      length: 70
      styles:
        - opacity: 0.2;
  vlines:
    - id: 0
      xpos: 50
      ypos: 59
      length: 36
      styles:
        - opacity: 0.2;
  states:
    - id: 0
      entity_index: 0
      xpos: 50
      ypos: 35
      styles:
        - font-size: 3em;
        - opacity: 0.9;
    - id: 1
      entity_index: 1
      xpos: 47
      ypos: 53
      styles:
        - font-size: 1.5em;
        - text-anchor: end;
    - id: 2
      entity_index: 2
      xpos: 53
      ypos: 53
      styles:
        - text-anchor: start;
        - font-size: 1.5em;
    - id: 3
      entity_index: 3
      xpos: 46
      ypos: 63
      styles:
        - text-anchor: end;
        - font-size: 1.5em;
    - id: 4
      entity_index: 4
      xpos: 53
      ypos: 63
      styles:
        - text-anchor: start;
        - font-size: 1.5em;
    - id: 5
      entity_index: 5
      xpos: 46
      ypos: 73
      styles:
        - text-anchor: end;
        - font-size: 1.5em;
    - id: 6
      entity_index: 6
      xpos: 53
      ypos: 73
      styles:
        - text-anchor: start;
        - font-size: 1.5em;
    - id: 7
      entity_index: 8
      xpos: 46
      ypos: 83
      styles:
        - text-anchor: end;
        - font-size: 1.5em;
    - id: 8
      entity_index: 9
      xpos: 53
      ypos: 83
      styles:
        - text-anchor: start;
        - font-size: 1.5em;
    - id: 9
      entity_index: 7
      xpos: 0
      ypos: 7
      styles:
        - text-anchor: start;
        - font-size: 1.2em;
  icons:
    - id: 0
      entity_index: 1
      xpos: 1
      ypos: 53
      align: start
      size: 1
  names:
    - id: 0
      entity_index: 0
      xpos: 50
      ypos: 95
      styles:
        - font-size: 1.2em;
    - id: 1
      entity_index: 1
      xpos: 15
      ypos: 45
      styles:
        - text-anchor: start;
        - font-size: 0.5em;
    - id: 2
      entity_index: 2
      xpos: 85
      ypos: 45
      styles:
        - text-anchor: end;
        - font-size: 0.5em;
    - id: 3
      entity_index: 7
      xpos: 0
      ypos: 12
      styles:
        - text-anchor: start;
        - font-size: 0.5em;
horseshoe_scale:
  min: 0
  max: 110000
  width: 3
color_stops:
  '0': green
  '1000': blue
  '2500': orange
  '4000': red
show:
  horseshoe_style: autominmax
card_mod:
  style: |
    ha-card {
      --ha-card-background: var(--card-background-color);
      color: var(--primary-color);
    }

```

bsafh, 2024-06-18:
well, it's not me who defined the grid layout, that's a straight copy from [JustGrabbingSomeIdeas](https://github.com/tomatensaus/DeyeSolarDesktop/blob/main/JustGrabbingSomeIdeas.md) and the YAML code for the dashboard over there ...

the empty card in https://github.com/tomatensaus/DeyeSolarDesktop/issues/5#issuecomment-2173177028 happens even in Mansonry layout

tomatensaus, 2024-06-21:
I had a user with another case like this. The dashboard would not render. Shift + F5 and then suddenly it renders.
I have not installed the latests HACS updates and mine still renders fine.

I did login remotely and confirmed that my browser showed the same issue.

I believe somewhere one of the hacs display components is causing this. Perhaps look at the browser log for error messages?

Also saw other people struggling with the same issue
https://community.home-assistant.io/t/lovelace-dashboard-not-loading-until-refresh/394876

tomatensaus, 2024-06-22:
Empty dashboard issue was resolved when the unknown entities were all fixed, strange indeed!

bsafh, 2024-06-25:
> The default layout "Masonry" is assumed.
> 
> If you are customising the layout with a grid layout then you need to check for yaml in the display definitions that might conflict with your layout. Like the example below.
> 
> ```
>         view_layout:
>         grid-area: g2
> ```
> 
> g2 might not be defined causing a failure to render

Nope, it's ignored in horizontal or mansory layout, removing it does not help

bsafh, 2024-06-25:
> Empty dashboard issue was resolved when the unknown entities were all fixed, strange indeed!

Nope, that does not help either.

I found the issue:

this particular `flex-horseshoe-card` definies 5 `entities`, but references `entity_index` number 0 to 9 in the "`states:`" section, i.e. many more entities than are actually defined.

The card shows up when either removing all references to non-existsting `entity_index` numbers or adding a few more (existing) `entities`.

So this particular card should be fixed in the template https://github.com/tomatensaus/DeyeSolarDesktop/blob/main/JustGrabbingSomeIdeas.md - as noone apart from the author can really guess whether there are `entities` missing or whether there are to many `states:` defined...

tomatensaus, 2024-06-25:
That sounds fixable. One other person reported this missing dashboard but when all the entities matched with the system it renders without any issues.

If I understand you correctly the issue is on the "Solar" flex-horshoe-card right?  Attached is a dashboard for reference

[image]

bsafh, 2024-06-25:
> If I understand you correctly the issue is on the "Solar" flex-horshoe-card right?

correct, yes

thanks!