tomatensaus/DeyeSolarDesktop #6: Number. vs select. in tabs "Time of Use" and "Time of Use 2"

bsafh, 2024-06-25:
I understand that the automations read the time settings from the inverter to the time data in those tabs and vice versa.

This assumes the values from the inverter are of type `number`. As I'm using the sunsyk addon (via modbus), those values are of type `select`.

Do I need the automations at all or can I just use the `select` entities directly in the dashboard?

If not, how do I convert `select` to `number` and back?

tomatensaus, 2024-06-25:
> I understand that the automations read the time settings from the inverter to the time data in those tabs and vice versa.

This part I can confirm, I take the raw number and convert it to a time object so that the editor on the UI validates the input. Any Change gets converted back to number and written again. I am planning to write a native date object in one of the new firmware versions that would allow the removal of the automation.

About your question:
Sorry but I have no idea about the select, I also assume it comes via MQTT, to me this seems like the wrong way to implement a date. I doubt you can change it and write it back.

https://www.home-assistant.io/integrations/select.mqtt/

bsafh, 2024-06-25:
Yes, it Codes via MQTT (the sunsyk deye addon communicates with HA via MQTT)

Looks ok, I guess

[image]
[image]

That's a high voltage hybrid inverter, the settings/config is a bit different from the low voltage inverters, there is no energy management mode selector in the Deye menues. It'ssplit up into several separate settings (Sell first/zero export to load/zero export to CT  + "allow sell" for "zero export to load" and separetly for "zero export to CT" + "export to grid load Limit in Watt")

I have two of those, and both need to have a battery of their own in order to work in Master-Slave Mode... you can't just connect both inverters to one battery like it's done for low voltage inverters.

tomatensaus, 2024-06-26:
I have several systems running the High Voltage version on my SmartDeyeDongle.

Could you please send some screenshots of the actual menus on the Deye. I developed my firmware on the Sunsynk 50 and there was very little difference, screens all look similar to the other low voltage inverters. I cater for 2 battery inputs (but combine them for the screen). 

I guess the difference is most people do not want to spend a week or more setting up their inverter dashboard before actually using it. How would the screen above set the time to 1:25  ? Is that possible?

bsafh, 2024-06-26:
I'll supply some Fotos.

1:25 is not directly possible, the `select` box is prefilled in 15 minutes Intervalls. But you can call a service to set any value you want: 

```
service: select.select_option
target:
  entity_id: select.my_entity
data:
  option: "example_option"
```
That's straight from [](https://www.home-assistant.io/integrations/select/).

I think there is even an option in rhe deye/sunsyk addon to set the desired intervall step. 15 minutes + the option Service is OK for me though.

bsafh, 2024-06-26:
@tomatensaus photos

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

bsafh, 2024-06-26:
Perhaps it's a software version thing, I have definitely seen different energy mode screens.

Anyway, works now over here.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

brashumza, 2024-07-09:
Good day

I’m also struggling with the time of use issue, how did you manage to resolve the issue.

brashumza, 2024-07-09:
> Yes, the `select` comes via MQTT (the sunsyk deye addon communicates with HA via MQTT)
> 
> Looks ok, I guess
> 
> ![Screenshot_20240625_232335_Home Assistant](https://private-user-images.githubusercontent.com/106388187/342936012-d06f4468-f7da-4efe-b67b-878e352dc20a.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1MjE0MDEsIm5iZiI6MTcyMDUyMTEwMSwicGF0aCI6Ii8xMDYzODgxODcvMzQyOTM2MDEyLWQwNmY0NDY4LWY3ZGEtNGVmZS1iNjdiLTg3OGUzNTJkYzIwYS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwOVQxMDMxNDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMWFlYzUzNGI4Yzg4NDc0ZTYwZTYwYjg5NTc4ZDExNTljMzI5ZTFjZGFmYjNmMTVhODhmMDhmZWYwMGM2ODdjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.bbbnIc4BeB4QfJF3gubK66eQpnmvDwZtBpkd2oboR1Y) ![Screenshot_20240625_232545_Home Assistant](https://private-user-images.githubusercontent.com/106388187/342936155-6d82badd-5346-4ac3-a56a-152a0d94451c.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA1MjE0MDEsIm5iZiI6MTcyMDUyMTEwMSwicGF0aCI6Ii8xMDYzODgxODcvMzQyOTM2MTU1LTZkODJiYWRkLTUzNDYtNGFjMy1hNTZhLTE1MmEwZDk0NDUxYy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwOVQxMDMxNDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NzMyYmY3OTBkYzRlNjc3MDI2Nzc5ODA3ZjY2MjczNjhmMTZkNWNkYzUyOWZjOTBiMGVlZjUyYWVlN2FkMGVjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.2ibUnEc5WbjMOsWfpdmnNuUs76dylAuj8sVPEjK3mJ4)
> 
> That's a high voltage hybrid inverter, the settings/config is a bit different from the low voltage inverters, there is no energy management mode selector in the Deye menues. It'ssplit up into several separate settings (Sell first/zero export to load/zero export to CT + "allow sell" for "zero export to load" and separetly for "zero export to CT" + "export to grid load Limit in Watt")
> 
> I have two of those, and both need to have a battery of their own in order to work in Master-Slave Mode... you can't just connect both inverters to one battery like it's done for low voltage inverters.

Can you kindly assist in how to manage to resolve the time of use issue