tomatensaus/DeyeSolarDesktop #7: Grid Charge Amps limited to 100A

genbankenergy, 2024-07-29:
the entity, number.deyeinvertermaster_grid_charge_current, is limited to a maximum of 100A. This is limiting the rate at which I can charge my batteries from the grid to the extent that I cannot charge from 20% to 100% during the three-hour Octopus Flux early morning charging period. My hardware is capable of charging at 150A. Please could this limitation be removed?

genbankenergy, 2024-07-30:
Thanks for the quick reply. I have an 8kW ecco EU inverter. (Sadly it lacks the internal 12 V supply to power your dongle. Otherwise the integration has worked flawlessly.)

I found the 120A limit in the octopus integration and understand how to modify it. The variable also seems to be limited to 100A at the esp32 firmware level which was the reason for raising the issue here 

Thanks again

tomatensaus, 2024-07-31:
I found the lines of code that needs to be corrected, I just need to study the impact on the 3Phase and High voltage code, I will roll out an update to the 4.5 Firmware asap(estimate it should take a day). Thank you for the feedback. I will leave this issue open and add instructions to do the upgrade.

tomatensaus, 2024-08-01:
I tested on my own 8kW inverter and was able to change the grid charge current with the changes made.

Steps to install the upgraded version (takes about 5 minutes):

Download this file https://smarthomeintegrations.co.za/demo14.bin

Go to the webpage on the module

At the bottom there is an upload button
Upload this file using the upload function.

It will load the demo. 
Then follow the normal process to upgrade the firmware to the latest version.
Since your inverter is registered you can follow this process to update to future versions.

Process shown in this part of the video https://youtu.be/1PDxWjiit5s?si=S3DuwkpZ8NSNqMV4&t=536

genbankenergy, 2024-08-01:
Thank you! I have successfully upgraded the firmware and the 100A limitation has been removed.

tomatensaus, 2024-08-01:
Thank you for the feedback

tomatensaus, 2024-10-07:
The upgrade instructions now on a separate issue. Closing this issue

closed 2024-10-07 by tomatensaus