tomatensaus/DeyeSolarDesktop #9: Upgrade the SmartDeyeDongle firmware

tomatensaus, 2024-09-27:
Steps to install the upgraded version of the firmware (takes about 5 minutes):

Note: Firmware is licensed per inverter, upgrades are free for the lifetime of the inverter.

Before you start:
You need to register your inverter serial number. 
Email me (HeinzMeulke ~at~ gmail.com  your inverter serial and also the order number.
If you have done this in the past there is no need to do it again, it will remain active.

Download this file https://smarthomeintegrations.co.za/demo14.bin    

Go to the web page on the module

At the bottom there is an upload button
Upload this file using the upload function.

It will load the demo.
Upgrade process shown in this part of the video https://youtu.be/1PDxWjiit5s?si=S3DuwkpZ8NSNqMV4&t=536

Follow the normal process to upgrade the firmware to the latest version.
This process will also work for future versions

martijnz43, 2026-11-11:
Hi Heinz,

How is it working with the updates after the dongle is registered? Will it auto update or do I need to flash it with a newer firmware? Can I check the current firmware somewhere?

Regards
Martijn

tomatensaus, 2026-11-14:
> Hi Heinz,
> 
> How is it working with the updates after the dongle is registered? Will it auto update or do I need to flash it with a newer firmware? Can I check the current firmware somewhere?
> 
> Regards Martijn

The process above loads the demo, the demo checks for the latest firmware and updates it. Once your inverter is registered you can install the firmware multiple times. The circuit was designed to outlast the inverter but if you replace the esp32 (V4) you can always flash it again.

If you install a different inverter and want to re-use the same dongle, you need to use the flash over USB method and select the erase option. (The dongle keeps certain information about the the inverter and will not operate correctly)