tomatensaus/DeyeSolarDesktop #17: Carte Homeassistant Maitre + Slave ne marche pas

connectic15, 2024-11-11:
Hello, I notice that only the solarDesktop.yaml card works for me. My system includes: 1 Master Inverter + 1 Slave Inverter in single phase. Can you make a functional .yaml file for a card including: 1 Master Inverter + 1 Slave Inverter in single phase

tomatensaus, 2024-11-12:
You should use the various dashboards applicable to your configuration, there is one for the Master, Slave, and Combined System view.

The Templates defined in template_MasterSlave_System.yaml will create the needed entities for the combined system view dashboard

connectic15, 2024-11-14:
Hello its working thank you very much for your help. I found an error in the code on the value: deyeinvertercombined_summary_day_grid_import_buy can you fix it? .

deyeinvertercombined_summary_day_grid_import_buy = DeyeInverterMaster Summary Day Grid Import(Buy) + DeyeInverterSlave Summary Day Grid Import(Buy)