tomatensaus/DeyeSolarDesktop #21: master+slave combined summary day grid import buy fix

inwookim, 2024-12-04:
There is an error in how master/slave systems need to calculate the import buy.  Change below:

```
    - name: "deyeinvertercombined_summary_day_grid_import_buy"
      state: "{{ (states('sensor.deyeinvertermaster_summary_day_grid_import_buy') |float(0) + (states('sensor.deyeinverterslave_summary_day_grid_import_buy') |float(0))) | round(1)}}"
      unit_of_measurement: "kWh"
      device_class: energy
      state_class: total_increasing
```

tomatensaus, 2024-12-10:
Thanks for the feedback!

I discovered that your limit to Load/Non-Essential/Allow Export also affects the reading.

In my experience the grid import is only measured by the master,. the Slave will have the exact same grid import amount. I tested this on two installs and it was true for both installs. By adding those two amounts together you get double the amount read by the official electricity meter (tested on limit to Load setting)

connectic15, 2024-12-22:
Hello HEINZ I have the same problem as INWOOKIM on 2 deye inverters of 5Kw in single-phase master/exclave mode. Photo in support of my electricity supplier meter EDF LINKY.

Can you tell me which file and lines to modify?

[image]
[image]

connectic15, 2024-12-23:
Pour les personne qui on le problème de l'import total d'achat d'énergie j'ai pu régler le problème en modifiant la ligne 157 du fichier template_MasterSlave_System.yaml :         
-     state: "{{ (states('sensor.deyeinvertermaster_summary_day_grid_import_buy') |float(0))}}"         
En modifiant la ligne 157 du fichier template_MasterSlave_System.yaml  par cette celle-ci : 
-     state: "{{ (states('sensor.deyeinvertermaster_summary_day_grid_import_buy') |float(0) + (states('sensor.deyeinverterslave_summary_day_grid_import_buy') |float(0))) | round(1)}}"

Merci INWOOKIM

healeydave, 2025-02-22:
I had the same issue, UK Users with 2x Sunsynk 5.5kW ECCO inverters paralleled. I have old mechanical meters still but I took photo's at 0845am yesterday and today and according to my units used, I consumed 36kWh. Without this modification, the combined dashboard was suggesting I had only used 17.4kWh

healeydave, 2025-02-23:
I have another query about the DATA. I have the 3x Dashboards installed (Master, Slave and Combined). On the Solar Power Usage Chart, why are the Battery Charge Percentages over 100% ?!

[image]

[image]

[image]