tomatensaus/DeyeSolarDesktop #22: Add Battery Efficiency entity to DeyeMasterInverter

Leventionz, 2024-12-23:
Hi,

Please could you add the entity for Battery Efficiency to DeyeMasterInverter  in ESPHome.

This is exposed by API so I assume its value is readable locally (it is also shown on inverter screen display to view battery health).

Thank you very much :-)

tomatensaus, 2024-12-31:
I tried looking for the SOH but could not find it in the Modbus Spec.
The only reference seems to be for using a Deye inverter with a Deye branded battery. I queried this on my system but only received 0. (I do not have a Deye branded battery)

I know this is supplied on the BMS CAN bus communication as part of the pylontech protocol.
If I can locate the correct register I will add it

Leventionz, 2024-12-31:
I use Sunsynk-L5.1 batteries https://www.sunsynk.org/ip65-battery I do not know if these are rebranded deye or not

okaymak, 2025-11-11:
I have Deye batteries. If you need anything from me to check, please let me know.

debug7878, 2025-11-11:
Hi tomatensaus

Don't know if this document might help. 

[ilide.info-modbus-single-phase-pr_2b076dcda6853bd899eb5a6cd456862d.pdf](https://github.com/user-attachments/files/23485165/ilide.info-modbus-single-phase-pr_2b076dcda6853bd899eb5a6cd456862d.pdf)

tomatensaus, 2025-11-28:
Thank you for the link. I was working off the V118 of this document, this appears to be V119. Very little changed. I am actually planning to install a 16kWh Deye in the next couple of months (currently out of stock at the supplier) and will see if I can get any new information from the inverter.
I tried the finding battery info from the registers and nothing returned any values, everything was reported as 0