tomatensaus/DeyeSolarDesktop #25: TOU Changes not carrying through Deye 5kw master/slave

davekennealy, 2025-06-09:
Time of use changes I make I HA are not being changed on inverter. If I make changes via the Deye dongle on rs232 they immediately reflect on the inverter.

tomatensaus, 2025-06-13:
Hi Dave,

The times on the time of use menu are populated with an automation that can be found in [smartdeyedongle_automations.yaml](https://github.com/tomatensaus/DeyeSolarDesktop/blob/main/packages/smartdeyedongle_automations.yaml)

The automation takes changes on the inverter and sets them in Home Assistant, it also takes the values that you type in the Home Assistant menu and applies them to the inverter to change the settings.

Automations allways needs to be copied to your automations.yaml file (in the packages folder they will be marked as read-only)

davekennealy, 2025-06-15:
Hi Heinz
The problem was with the inverters.I asked Deye Technical in China what the problem was and they immediately went in through the wifi dongles and updated the firmware and now working perfectly.
Really enjoying the system tThanks again.
Dave

tomatensaus, 2025-06-23:
I know that the Time Of Use settings were added around 2021 so if your inverter still had old firmware then an upgrade should fix most issues related to the ToU. Thank you for the feedback @davekennealy