tomatensaus/DeyeSolarDesktop #27: Writing to charge state - will not accept None

milsky99, 2025-09-15:
Loving the integration and have been up and running with the loadshedding automations being a blast

My only challenge was that I wanted to be able to amend my work schedule outside of scenes and have my own helpers to do this that would then write the values for both loadshedding as well as predicted solar.

I disabled the scene switcher and scenes 
Setup individual helpers for each timezone for SOC, charge selection and Energy model for 5 loadshedding stages and 3 weather stages where the weather stages would only be applied if there was no loadshedding stage. (ie not stage 0 or no loadshedding)
Then tried to setup an automation to write the relevant entities depending on these 8 situations.

I have made it a fair way and am able to write the SOCs but not the charge selection, well better put I can write Grid, but it is not accepting the "None" status, I do not have a generator so the other attributes are not relevant to me

What am I missing here ?

Help would be greatly appreciated.

Oh and its worth noting that I originally tried to set this up with the scenes and a modified automation that behaved like the original scenes and scene switcher however cannot get an automation or script to write the helper values to the scene values.

tomatensaus, 2025-09-16:
You should be using the right construct to set the value. Here is an example I used for setting up cheap charge cycles on eastern Europe pricing

[image]

milsky99, 2025-09-16:
Final comment so you have all the data, here are the helper values for just the stormy setting

  charge_stormy_tz1:
    name: "Stormy • TZ1 Charge"
    options: ["None","Grid"]
  charge_stormy_tz2:
    name: "Stormy • TZ2 Charge"
    options: ["None","Grid"]

You can see these are the same as the attributes you provided for timezone 1 charge

tomatensaus, 2025-10-20:
4.6 Is the latest version at the time of writing this comment.

No it would not be due to slow updates. Data that is unlikely to update frequently is fetched on a slower schedule to keep the batches of data smaller.

The only think I can think of is the inverter firmware. There were some issues with 2021 and older firmware, specifically on the time of use not updating. It is seldom that I encounter inverters with old firmware these days. I updated my deye at the beginning of 2023 and have not had any issues.

As a last resort, set the value via the HA dashboard and look at the debug messages on the web interface on the dongle itself